espressif/esp-skainet #139: waveshare ESP32-S3-DEV-KIT-N8R8 crashes after wake word detection

Blade19888, 2024-08-31:
Hello.
i am trying to make work the "en_speech_commands_recognition" example, but when the wake word detect the watchdog resets the esp, so i disable the watchdog, and now when detect the wake word stays still..

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1918
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe5c
load:0x403cb700,len:0x303c
entry 0x403c893c
I (27) boot: ESP-IDF v5.3 2nd stage bootloader
I (27) boot: compile time Aug 31 2024 22:13:33
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 factory          factory app      00 00 00010000 00200000
I (77) boot:  1 model            Unknown data     01 82 00210000 0050c000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=bc4ach (771244) map
I (214) esp_image: segment 1: paddr=000cc4d4 vaddr=3fc9b700 size=03b44h ( 15172) load
I (222) esp_image: segment 2: paddr=000d0020 vaddr=42000020 size=68770h (427888) map
I (287) esp_image: segment 3: paddr=00138798 vaddr=3fc9f244 size=01cd4h (  7380) load
I (289) esp_image: segment 4: paddr=0013a474 vaddr=40378000 size=136e4h ( 79588) load
I (316) boot: Loaded app from partition at offset 0x10000
I (316) boot: Disabling RNG early entropy source...
I (328) octal_psram: vendor id    : 0x0d (AP)
I (328) octal_psram: dev id       : 0x02 (generation 3)
I (328) octal_psram: density      : 0x03 (64 Mbit)
I (333) octal_psram: good-die     : 0x01 (Pass)
I (338) octal_psram: Latency      : 0x01 (Fixed)
I (344) octal_psram: VCC          : 0x01 (3V)
I (349) octal_psram: SRF          : 0x01 (Fast Refresh)
I (355) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (360) octal_psram: BurstLen     : 0x01 (32 Byte)
I (366) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (372) octal_psram: DriveStrength: 0x00 (1/1)
I (378) MSPI Timing: PSRAM timing tuning index: 5
I (383) esp_psram: Found 8MB PSRAM device
I (387) esp_psram: Speed: 80MHz
I (391) cpu_start: Multicore app
I (716) esp_psram: SPI SRAM memory test OK
I (725) cpu_start: Pro cpu start user code
I (725) cpu_start: cpu freq: 240000000 Hz
I (725) app_init: Application information:
I (728) app_init: Project name:     speech_commands_recognition
I (735) app_init: App version:      v5.3
I (739) app_init: Compile time:     Aug 31 2024 22:13:03
I (745) app_init: ELF file SHA256:  543bbe96d...
I (750) app_init: ESP-IDF:          v5.3
I (755) efuse_init: Min chip rev:     v0.0
I (760) efuse_init: Max chip rev:     v0.99
I (765) efuse_init: Chip rev:         v0.2
I (770) heap_init: Initializing. RAM available for dynamic allocation:
I (777) heap_init: At 3FCA1D58 len 000479B8 (286 KiB): RAM
I (783) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (789) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (795) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (803) spi_flash: detected chip: gd
I (807) spi_flash: flash io: qio
W (811) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (822) sleep: Configure to isolate all GPIO pins in sleep state
[... 109 lines cut ...]
phrase id:10, command id:10, str:T k N   e F   j c   T m V m
phrase id:11, command id:11, str:M d K   M m   c   T m
phrase id:12, command id:12, str:M d K   M m   c   K n F m
phrase id:13, command id:13, str:T k N   n N   j c   L i T
phrase id:14, command id:14, str:T k N   e F   j c   L i T
phrase id:15, command id:15, str:p d N q   j c   K c L k   T o   R f D
phrase id:16, command id:16, str:p d N q   j c   K c L k   T o   G R m N
phrase id:17, command id:17, str:T k N   n N   e L   j c   L i T S
phrase id:18, command id:18, str:T k N   e F   e L   j c   L i T S
phrase id:19, command id:19, str:T k N   n N   j c   f R   K c N D g s c N k
phrase id:20, command id:20, str:T k N   e F   j c   f R   K c N D g s c N k
phrase id:21, command id:21, str:S f T   j c   T f M P R c p k   T o   S g K S T m N   D g G R m Z
phrase id:22, command id:22, str:S f T   j c   T f M P R c p k   T o   S f V c N T m N   D g G R m Z
phrase id:23, command id:23, str:S f T   j c   T f M P R c p k   T o   d T m N   D g G R m Z
phrase id:24, command id:24, str:S f T   j c   T f M P R c p k   T o   N i N T m N   D g G R m Z
phrase id:25, command id:25, str:S f T   j c   T f M P R c p k   T o   T W f N T m   D g G R m Z
phrase id:26, command id:26, str:S f T   j c   T f M P R c p k   T o   T W f N T m   W c N   D g G R m Z
phrase id:27, command id:27, str:S f T   j c   T f M P R c p k   T o   T W f N T m   T o   D g G R m Z
phrase id:28, command id:28, str:S f T   j c   T f M P R c p k   T o   T W f N T m   v R m   D g G R m Z
phrase id:29, command id:29, str:S f T   j c   T f M P R c p k   T o   T W f N T m   F e R   D g G R m Z
phrase id:30, command id:30, str:S f T   j c   T f M P R c p k   T o   T W f N T m   F i V   D g G R m Z
phrase id:31, command id:31, str:S f T   j c   T f M P R c p k   T o   T W f N T m   S g K S   D g G R m Z
------------detect start------------
WAKEWORD DETECTED
AFE_FETCH_CHANNEL_VERIFIED, channel index: 2

Blade19888, 2024-08-31:
Thank you for any help you can provide.

sun-xiangyu, 2024-09-02:
@Blade19888 
We do not support this ESP32-S3-DEV-KIT-N8R8 board by default. you will need to add the board according to the template [here](https://github.com/espressif/esp-skainet/blob/master/components/hardware_driver/boards/esp32s3-korvo-2/bsp_board.c).