espressif/esp-skainet #140: example里的chinese_tts播放有奇怪的尾音

ysni, 2024-09-03:
按readme说明编译烧录到esp32s3-korvo-2板子上后,播放完语句会出现哒哒哒哒的不间断声音,是什么问题呢

sun-xiangyu, 2024-09-03:
可以试试在播放完成后添加
```i2s_zero_dma_buffer(0)```

ysni, 2024-09-03:
> 可以试试在播放完成后添加 `i2s_zero_dma_buffer(0)`

会报错,版本冲突了i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver,我的是idf5.0.6

ysni, 2024-09-03:
> > 可以试试在播放完成后添加 `i2s_zero_dma_buffer(0)`
> 
> 会报错,版本冲突了i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver,我的是idf5.0.6

这里用到是skainet里封装的esp_audio_play,不是adf框架里的esp_audio_play打开语音,不能用adf里的esp_audio_stop来关掉语音

sun-xiangyu, 2024-09-03:
> > 可以试试在播放完成后添加 `i2s_zero_dma_buffer(0)`
> 
> 会报错,版本冲突了i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver,我的是idf5.0.6

你也可以试一下下面的方法，直接在结构体中设置 auto_clear = true
https://github.com/espressif/esp-skainet/blob/8eac0bc3e46462f0131486b26b8a294f401d8ef1/components/hardware_driver/boards/esp32p4-function-ev/bsp_board.c#L125

你上面的报错应该是 I2S 新旧版本互用产生的

sun-xiangyu, 2024-09-03:
> > > 可以试试在播放完成后添加 `i2s_zero_dma_buffer(0)`
> > 
> > 
> > 会报错,版本冲突了i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver,我的是idf5.0.6
> 
> 这里用到是skainet里封装的esp_audio_play,不是adf框架里的esp_audio_play打开语音,不能用adf里的esp_audio_stop来关掉语音

是的，esp-skainet 里没有 esp_audio_stop

howelljiang, 2025-04-26:
> > > 可以试试在播放完成后添加 `i2s_zero_dma_buffer(0)`
> > 
> > 
> > 会报错,版本冲突了i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver,我的是idf5.0.6
> 
> 你也可以试一下下面的方法，直接在结构体中设置 auto_clear = true
> 
> [esp-skainet/components/hardware_driver/boards/esp32p4-function-ev/bsp_board.c](https://github.com/espressif/esp-skainet/blob/8eac0bc3e46462f0131486b26b8a294f401d8ef1/components/hardware_driver/boards/esp32p4-function-ev/bsp_board.c#L125)
> 
> Line 125 in [8eac0bc](/espressif/esp-skainet/commit/8eac0bc3e46462f0131486b26b8a294f401d8ef1)
> 
>  chan_cfg.auto_clear = true; 
> 你上面的报错应该是 I2S 新旧版本互用产生的

 chan_cfg.auto_clear = true;  加这个有用。感谢大佬。