espressif/esp-skainet #145: voice communicate 例程中添加 VAD 支持

iysheng, 2024-12-31:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

请问 voice communicate  中是否可以添加 VAD (only for speech recognition) 功能检测是否存在人声？

### Use cases

我想实现不需要唤醒词，当检测到有人说话时就执行一些操作（比如回放说话的内容）。

shenxiaozheng, 2024-12-31:
You can open nsnet2 to eliminate noise other than human voices. Then use VAD. Of course, our new VAD has excellent performance, please stay tuned.

iysheng, 2024-12-31:
> You can open nsnet2 to eliminate noise other than human voices. Then use VAD. Of course, our new VAD has excellent performance, please stay tuned.
[image]

根据网线描述，我看 vad只在 speech recognition 中配置是否开启，voice communication 也可以配置开启 vad_init ?

shenxiaozheng, 2024-12-31:
voice communication 也可以配置不可以开启 vad_init

iysheng, 2024-12-31:
> voice communication 也可以配置不可以开启 vad_init

你想表达的是不是 voice communication 也可以配置为开启 vad_init ?😂

shenxiaozheng, 2025-01-02:
voice communication 不可以开启 vad_init

iysheng, 2025-01-02:
> You can open nsnet2 to eliminate noise other than human voices. Then use VAD. Of course, our new VAD has excellent performance, please stay tuned.

那在 voice communication 不可以开启 vad_init 的前提下，如 issue 所描述，现在有没有什么好的建议？

sun-xiangyu, 2025-01-02:
@iysheng , 不是不可以开启，是现在还没有支持。
有一个问题，你希望在降噪算法前面加VAD,还是降噪算法后面加VAD

iysheng, 2025-01-02:
> @iysheng , 不是不可以开启，是现在还没有支持。
> 有一个问题，你希望在降噪算法前面加VAD,还是降噪算法后面加VAD

我更关注降噪之后的内容，更偏向降噪之后加 VAD，可以有 API 让我判断是否有有效的人声。

sun-xiangyu, 2025-05-06:
The new vad: https://github.com/espressif/esp-skainet/tree/master/examples/voice_activity_detection