espressif/esp-skainet #147: 我参考 examples 文件夹下的例程，如何时间 mp3 格式音频播放?是否有相关参考？

iysheng, 2025-01-11:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

MP3 音频格式播放参考

### Use cases

我想尝试直接播放 MP3 格式音频而不是 PCM 格式的，是否有相关参考?

### Alternatives

MP3 音频播放参考。

sun-xiangyu, 2025-01-13:
可以参考esp-adf的例子: https://github.com/espressif/esp-adf/tree/master/examples/player

iysheng, 2025-02-18:
我想实现动态调节采样率，写了一个 mp3 解码的回调函数，内容如下，

[image]

但是执行的时候，报错如下
```
E (104218) i2s_common: i2s_channel_disable(1107): the channel has not been enabled yet
I (104228) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (104228) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:576 mask:1
I (104248) Adev_Codec: Open codec device OK
```