espressif/esp-skainet #149: 使用esp_audio_play()方式播放声音，是否可以实现阻塞播放和非阻塞播放

monskate, 2025-01-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

实现非阻塞播放声音，播放声音时可以打断

### Use cases

使用chinese_tts例程中的方式转换文字，怎样实现可以打断他播放声音，或者实现非阻塞播放声音

sun-xiangyu, 2025-01-24:
打断后，还需要继续之前的播放吗？
- 如果不需要，可以直接使用 esp_tts_stream_reset(tts_handle); 就可以清空 
- 如果需要，你需要将tts 的结果保存到一个ringbuffer中，然后自定义播放和结束

monskate, 2025-01-24:
> 打断后，还需要继续之前的播放吗？
> 
> * 如果不需要，可以直接使用 esp_tts_stream_reset(tts_handle); 就可以清空
> * 如果需要，你需要将tts 的结果保存到一个ringbuffer中，然后自定义播放和结束

好的，已解决，如果我想使用cn_speech_commands_recognition例程获取麦克风输入的音频并保存，应该怎么实现，使用的esp32s3-korvo-2的板子

sun-xiangyu, 2025-01-24:
可以参考[ examples/wake_word_detection](https://github.com/espressif/esp-skainet/blob/a7fdf56add0c81f78a5705ae2710e5091b9e6234/examples/wake_word_detection/main/main.c#L37)