espressif/esp-skainet #151: error: implicit declaration of function 'esp_cpu_get_cycle_count'

dony71, 2025-01-24:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I use ESP-IDF v.4.4.2 and try to compile "esp-skainet/examples/en_speech_commands_recognition/"
Compilation stop below .... Anybody help me what's wrong?

### Actual behavior (suspected bug)

I use ESP-IDF v.4.4.2 and try to compile "esp-skainet/examples/en_speech_commands_recognition/"
Compilation stop below .... Anybody help me what's wrong?

### Error logs or terminal output

```shell
C:/Project/esp-skainet/components/perf_tester/wn_perf_tester.c: In function 'fetch_task':
C:/Project/esp-skainet/components/perf_tester/wn_perf_tester.c:281:14: error: implicit declaration of >function 'esp_cpu_get_cycle_count'; did you mean 'esp_cpu_get_ccount'? [-Werror=implicit-function->declaration]
         c0 = esp_cpu_get_cycle_count();
              ^~~~~~~~~~~~~~~~~~~~~~~
              esp_cpu_get_ccount
cc1.exe: some warnings being treated as errors
[3/20] Building C object esp-idf/perf_tester/CMakeFiles/__idf_perf_tester.dir/mn_perf_tester.c.obj
FAILED: esp-idf/perf_tester/CMakeFiles/__idf_perf_tester.dir/mn_perf_tester.c.obj
```

### Steps to reproduce the behavior

VS code with ESP-IDF extension v.4.4.2

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

other (details in Additional context)

### Additional context

VS code with ESP-IDF extension v.4.4.2

sun-xiangyu, 2025-01-24:
The issue you're encountering is due to the IDF version being too low. You can directly upgrade to v5.0 or higher. If you prefer not to upgrade, you can simply comment out the relevant code, as it is just used for performance testing.

dony71, 2025-01-24:
> The issue you're encountering is due to the IDF version being too low. You can directly upgrade to v5.0 or higher. If you prefer not to upgrade, you can simply comment out the relevant code, as it is just used for performance testing.

I try to build using latest ESP-IDF v5.4, now I got different error. So which ESP-IDF can build this well?

C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c: In function 'led_Task':
C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c:71:56: error: passing argument 3 of 'led_strip_new_rmt_device' from incompatible pointer type [-Wincompatible-pointer-types]
   71 |     led_strip_new_rmt_device(&led_config, &rmt_config, &strip);
      |                                                        ^~~~~~
      |                                                        |
      |                                                        struct led_strip_t ***
In file included from C:/Project/esp-skainet/examples/en_speech_commands_recognition/managed_components/espressif__led_strip/include/led_strip.h:10,
                 from C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c:59:
C:/Project/esp-skainet/examples/en_speech_commands_recognition/managed_components/espressif__led_strip/include/led_strip_rmt.h:49:136: note: expected 'struct led_strip_t **' but argument is of type 'struct led_strip_t ***'      
   49 | esp_err_t led_strip_new_rmt_device(const led_strip_config_t *led_config, const led_strip_rmt_config_t *rmt_config, led_strip_handle_t *ret_strip);
      |
         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
In file included from C:/Users/donyo/esp/v5.4/esp-idf/components/esp_hw_support/include/esp_intr_alloc.h:12,      
                 from C:/Users/donyo/esp/v5.4/esp-idf/components/esp_hw_support/include/esp_cpu.h:20,
                 from C:/Users/donyo/esp/v5.4/esp-idf/components/esp_hw_support/include/spinlock.h:11,
                 from C:/Users/donyo/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:49,
                 from C:/Users/donyo/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from C:/Users/donyo/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c:11:
C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c:76:37: error: passing argument 1 of 'led_strip_clear' from incompatible pointer type [-Wincompatible-pointer-types]
   76 |     ESP_ERROR_CHECK(led_strip_clear(strip));
      |                                     ^~~~~
      |                                     |
      |                                     struct led_strip_t **
C:/Users/donyo/esp/v5.4/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
C:/Project/esp-skainet/examples/en_speech_commands_recognition/managed_components/espressif__led_strip/include/led_strip.h:96:46: note: expected 'led_strip_handle_t' {aka 'struct led_strip_t *'} but argument is of type 'struct led_strip_t **'
   96 | esp_err_t led_strip_clear(led_strip_handle_t strip);
      |                           ~~~~~~~~~~~~~~~~~~~^~~~~
C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c:78:45: error: passing argument 1 of 'led_strip_set_pixel' from incompatible pointer type [-Wincompatible-pointer-types]
   78 |         ESP_ERROR_CHECK(led_strip_set_pixel(strip, j, 50, 50, 50));
      |                                             ^~~~~
      |                                             |
      |                                             struct led_strip_t **
C:/Users/donyo/esp/v5.4/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
C:/Project/esp-skainet/examples/en_speech_commands_recognition/managed_components/espressif__led_strip/include/led_strip.h:35:50: note: expected 'led_strip_handle_t' {aka 'struct led_strip_t *'} but argument is of type 'struct led_strip_t **'
   35 | esp_err_t led_strip_set_pixel(led_strip_handle_t strip, uint32_t index, uint32_t red, uint32_t green, uint32_t blue);
      |                               ~~~~~~~~~~~~~~~~~~~^~~~~
C:/Project/esp-skainet/examples/en_speech_commands_recognition/main/speech_commands_action.c:81:39: error: passing argument 1 of 'led_strip_refresh' from incompatible pointer type [-Wincompatible-pointer-types]
   81 |     ESP_ERROR_CHECK(led_strip_refresh(strip));
      |                                       ^~~~~
      |                                       |
      |                                       struct led_strip_t **
C:/Users/donyo/esp/v5.4/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
[... cut ...]
C:/Project/esp-skainet/examples/en_speech_commands_recognition/managed_components/espressif__led_strip/include/led_strip.h:85:48: note: expected 'led_strip_handle_t' {aka 'struct led_strip_t *'} but argument is of type 'struct led_strip_t **'
   85 | esp_err_t led_strip_refresh(led_strip_handle_t strip);
      |                             ~~~~~~~~~~~~~~~~~~~^~~~~
[1196/1223] Building C object esp-idf/sr_ringbuf/CMakeFiles/__idf_sr_ringbuf.dir/lock.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Users\donyo\.espressif\tools\ninja\1.12.1\ninja.exe" terminated with exit code: 1.