espressif/esp-skainet #153: how could disable these print.

iysheng, 2025-02-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

how could disable these output:
```
model_run_step2 in
model_run_step3 in
model_run_step1 in
model_run_step2 in
model_run_step3 in
model_run_step1 in
model_run_step2 in
model_run_step3 in
model_run_step1 in
model_run_step2 in
model_run_step3 in
model_run_step1 in
```

sun-xiangyu, 2025-02-18:
please update esp-sr to V2.0

iysheng, 2025-02-18:
> please update esp-sr to V2.0

yeah，after I update with upstream，no these print now.

closed 2025-02-18 by iysheng